ShpakovDmitry/My-old-embedded-projects #18: Remap button ports to another ports.

ShpakovDmitry, 2020-02-11:
Button ports intersect with ISP ports. On every button there is capacitor connected for de-bounce
purposes. This will disturb communication with programming device over SPI.

https://github.com/ShpakovDmitry/My-old-projects/blob/a6789ca00a900fd6a1314b799cd6dcc4e96271ce/embedded/EscapeROOM/radio_morse_panel/radio_morse_panel.c#L161-L163

ShpakovDmitry, 2020-02-11:
Suggested remap according to schematics: [radio_morse_panel.pdf](embedded/EscapeROOM/radio_morse_panel/schematics/radio_morse_panel.pdf)

```c
unsigned char button1 = ~(1 << PB0); 
unsigned char button2 = ~(1 << PB1); 
unsigned char button3 = ~(1 << PB2); 
```

krisjans, 2020-02-12:
How about leaving those buttons where they are? Instead remove those capacitors AND write proper digital noise filter?

ShpakovDmitry, 2020-02-12:
Agree, digital noise filter would be better solution, which lead to no extra components.
Then it is possible to connect buttons to SPI ports to reduce port usage.
Suggested remap would be:
```c
unsigned char button1 = ~(1 << PB5); 
unsigned char button2 = ~(1 << PB6); 
unsigned char button3 = ~(1 << PB7); 
```

krisjans, 2020-02-12:
I think leave as-is for now. Let's limit changes to sw for now.
Imagine if this hardware was deployed to 10000 customers and the only option is to change sw.

ShpakovDmitry, 2020-02-12:
Ok, this is the best example. I like it.
Leaving as is.

closed 2020-02-12 by ShpakovDmitry