ShpakovDmitry/My-old-embedded-projects #21: Newer give the one net two names.

krisjans, 2020-02-12:
[image]
MISO and BUTTON_1. This hides signal co-dependency. If all I care is ISP connector, then I will not notice that one of the signals are re-used by button. And if I search by BUTTON_1 in pdf, I will not find connection to ISP connector.
Is there any reason for doing so? Any schematics drawing practices? I might be wrong after all - I'm not an electronic engineer.
I strongly advocate towards naming net by it's primary function - in this case BUTTON_1. MISO signal for ISP is just meta - needed only during programming.
[embedded/EscapeROOM/radio_morse_panel/schematics/radio_morse_panel.pdf](
https://github.com/ShpakovDmitry/My-old-projects/blob/master/embedded/EscapeROOM/radio_morse_panel/schematics/radio_morse_panel.pdf)

ShpakovDmitry, 2020-02-12:
Actually, i can not tell about existing "standard" practice naming nets, only if: one net - one name. But here i wanted to show that pin has two functionalities. But as you pointed out, focusing on primary functionality is the key.

ShpakovDmitry, 2020-02-13:
Also if net has two labels in schematics it will have only one label at PCB routing, ordinary the first added.(True for KiCAD)

PR #22 (2020-02-13): left one label - one net; schematics updated
Updated schematics.

Closes #21 

/cc @krisjans
closed 2020-02-13 by ShpakovDmitry