ShpakovDmitry/My-old-embedded-projects #23: Actually there is eight buttons not three. Suppose to change schematics.

ShpakovDmitry, 2020-02-17:
After building the prototype i realized that three input buttons is too simple to activate the speaker. When looking at the code:
https://github.com/ShpakovDmitry/My-old-projects/blob/6e86ebe9eb74b7c9b303172c605751fa2a4893e7/embedded/EscapeROOM/radio_morse_panel/radio_morse_panel.c#L179-L180
this tells that microcontroller monitors all the eight PORTB inputs. 
And then i remembered that there actually were eight buttons.
I suggest to change schematics where all eight buttons are presented.

krisjans, 2020-02-17:
Great job!
You are right. Eight buttons. Makes more sense.
And now when you pointed that out I can understand why I missed that at first.
https://github.com/ShpakovDmitry/My-old-projects/blob/6e86ebe9eb74b7c9b303172c605751fa2a4893e7/embedded/EscapeROOM/radio_morse_panel/radio_morse_panel.c#L161-L185
- First - only three buttons defined (button1-button3). BTW - they are defined in a supper expensive way - instead of #define you use variables. RAM variables - there is no "const" statement so compiler can't move them to FLASH memory or optimize it away. Consider enum.
- Second - during check for button release condition those variables are not used.
- Third - local variable "data" is used to cache port input (PINB). Generic name that hides link between PINB and button variables (button1-button3).

P.S. Lines 179-180 deserve their own special place in hell for several reasons. How many reasons do you see now?
P.S.2. And yes - me too -  I used to write code like that. To be honest - I did even worse.

ShpakovDmitry, 2020-02-18:
Thank you!
About
- First - agree this is waste of memory and optimization
- Second - that is

ShpakovDmitry, 2020-02-18:
Accidently closed

krisjans, 2020-02-18:
> actually does not do what it meant to do - check if only one button is pressed

... and I thought that intention is to check if **any button** is pressed. Therefore my proposed fix was to replace those tow lines with an else statement:
```
 	if(data == 0xff) 
 	{ 
 		return; 
 	} 
 	else // "else in this case is equal to !(data & 0x01) || !(data & 0x02) || !(data & 0x04) || !(data & 0x08) || !(data & 0x10) || !(data & 0x20) || !(data & 0x40) || !(data & 0x80)
 	{ 
```
Even my proposed "else" statement is redundant - just use one "if" statement:
```
 	if(data != 0xff) 
 	{ 
```
But such change would brake existing "game logic".

> check if only one button is pressed

Check _"if only one button is pressed"_ can't be expressed in one "if" statement in a sane way.
My proposal would be something like this:
```
#define BTN_0 PB0
#define BTN_0_MASK (1 << BTN_0)
#define BTN_1 PB1
#define BTN_1_MASK (1 << BTN_1)
#define BTN_2 PB2
#define BTN_2_MASK (1 << BTN_2)
#define BTN_3 PB3
#define BTN_3_MASK (1 << BTN_3)
#define BTN_4 PB4
#define BTN_4_MASK (1 << BTN_4)
#define BTN_5 PB5
#define BTN_5_MASK (1 << BTN_5)
#define BTN_6 PB6
#define BTN_6_MASK (1 << BTN_6)
#define BTN_7 PB7
#define BTN_7_MASK (1 << BTN_7)
#define BTN_MASK (BTN_0_MASK | BTN_1_MASK | BTN_2_MASK | BTN_3_MASK \
        | BTN_4_MASK | BTN_5_MASK | BTN_6_MASK | BTN_7_MASK)

ISR(PCINT_vect)
{
	static unsigned char passwordIndex = 0;

        unsigned char scanButtons = (~PINB) & BTN_MASK;

        // Insert digital noise filter here

        if (scanButtons == 0) {
                return;
        }
        if ( (scanButtons == BTN_0_MASK)
                || (scanButtons == BTN_1_MASK)
                || (scanButtons == BTN_2_MASK
                || (scanButtons == BTN_3_MASK
                || (scanButtons == BTN_4_MASK
                || (scanButtons == BTN_5_MASK
                || (scanButtons == BTN_6_MASK
                || (scanButtons == BTN_7_MASK) )
        {
                // Store button code into input buffer or do check against password  
        }
        else {
                passwordIndex = 0;
        }
}
```
But I'm not happy with my proposal. That 8 line long "if" statement also deserves special place in hell.

The whole approach must be different. Something like this, perhaps:

- get input from GPIOs (during timer and/or gpio-change-event interrupt) and pass it to noise-filter.
- the filter detects button-down and button-up events and store those events in buffer.
- in main program get button events from that buffer and compare with constant password string.

krisjans, 2020-02-18:
> Please, tell what flaws do you see in this code?

some more flaws:

- use of negative logic. For example "if **not Sunday** then **do work** otherwise **eat cake**" uses negative logic. It is easier to read positive logic: "if **Sunday** then **eat cake** otherwise **do work**".
- use of super-complex expressions. This is direct violation of "code must be easy to read" mantra.  And sometimes it signals that you are trying to do too many things at once. If you can't get rid of such complex check (and yes, sometimes you just can't), then consider moving it to static inline bool function with meaningful name.
- cryptic variable name ("data"). In some cases, "data" as variable name is OK - if you copy, send, receive or parse data blocks. But not here- this is scanCode or portValue or just "input". If you can't find name that fits, then better use one letter variable (for example "a") - especially if it's life span is short.
- hard-coded magic **numbers** ( not magic constants).
- use of \ at the end of line in C code. \ at the end of line is only needed to define long macros with #define. Normal C/C++ code allows white-space (space, tabs, newline) to be added almost anywhere for formatting purposes.

ShpakovDmitry, 2020-02-18:
Great! These are very valuable comments for me. Thank you!
Now i look at the code one more time and can not remember for what purpose actually these lines were meant to be. At the end i will agree with you that lines 179-180 of code, checked that if **any** button was pressed.

ShpakovDmitry, 2020-02-18:
After button additions to schematics, i think now there is a mess with button net labeling.

This happened because button1, button2 and button3 have **"special"** place in code.
Only those button press sequence matters, other buttons only resets entering sequence.

ShpakovDmitry, 2020-02-18:
When this code were created years ago, i have not any thoughts about code re-usability, even more about good programming style or self-documenting code, i only cared about successful device working state.

PR #25 (2020-02-18): Issue 23 solve.
Updated schematics.

Added missing buttons. Now there are eight of them.

closes #23 

/cc @krisjans
closed 2020-02-18 by ShpakovDmitry