ShpakovDmitry/My-old-embedded-projects #24: Write a proper digital filter for directly connected buttons.

krisjans, 2020-02-17:
https://github.com/ShpakovDmitry/My-old-projects/blob/6e86ebe9eb74b7c9b303172c605751fa2a4893e7/embedded/EscapeROOM/radio_morse_panel/radio_morse_panel.c#L159-L239

ShpakovDmitry, 2020-02-17:
I have the following idea regarding digital filter.
Here should be used low-pass-filter with time constant [image] much bigger than bounce time [image] when key is pressed or released, but much lower than "regular" press duration time [image].
Simple **IIR** low-pass filter equation would be the following:
[image]
where:
[image]
[image] - time between button state read,
[image] - current actual state of the button,
[image] - filtered button state,
[image] - previous filtered button state.

Also, using this, it is possible to introduce hysteresis.

I know, this demands floating-point arithmetic, but in this case it is possible to use fixed-point arithmetic.

I would create TIMER OVF ISR which every time [image] reads button actual state [image]
calculates and updates button filtered state [image] and according to hysteresis send "signal" to method which takes an action on button state change.

#### Code example:
```c
ISR(TIMER_OVF){
  // read actual button state through PINB
  // set the flag
  STATE_UPDATED_flag = true;
}

void UpdateButtonState(void){
  if(STATE_UPDATED_flag){
    STAE_UPDATED_flag = false;
    // calculate new filtered button states
    // update button state using hysteresis and filtered states
    // if any button state changed put it in FIFO or
    // better in cyclic buffer for further button sequence analysis
  }
}

void CheckValidPressSequence(void){
  // ...
  if(correct)
    PlayMorsePhrase();
  else
    SwitchOnRedLed();
}

```

This will change the code a lot, but i think that according to Issue #14, code anyway should be changed a lot.

Also this IIR filter implementation maybe more complicated than waiting while time elapsed from last de-bounce event is larger than de-bounce duration.

krisjans, 2020-02-19:
You might want to fix issue #14 before this one. First make ISR fast and non-blocking and after that introduce proper filter.

ShpakovDmitry, 2020-02-24:
@krisjans 

Please, can you give an advice on the following.
Maybe it is better to create new branch called `dev`, with infinite lifetime, and in future merge branches that resolves issues into `dev` branch? And when everything works fine merge it into `master` branch and continue work on `dev` branch.
This is due to the following. I think there might be the situation when issue is successfully solved, code may become, lets say, in `transition` state, when it is not fully working and demands new feature to be added. So, we develop code in `dev` branch until it starts to work and then `release` it by merging into `master` branch. Thus it is possible to merge into `master` branch only in situations when code is fully functional. And thus, it is possible to access to functional code(`master`) at any time.
Correct me, please, if i am wrong regarding development flow.

krisjans, 2020-02-24:
@ShpakovDmitry
Yes, just like coding guidelines - it is about taste and convenience - use whatever you like.

But - please take a look how other teams deals with this problem. Google for "git branch strategy" - you will find interesting articles, diagrams and videos about this topic.

In my opinion, general "rule" is that "master" is the main branch where development happens. Developers either commit directly to the master OR create feature branch for every bug or feature  and when it is done merge it back to master. Inside feature branch you may solve several issues and even have non-working or unstable states.
Release is either "tag" in master or "release branch" from master.

P.S. I see that you create branch names  to match issue numbers. But this makes things pretty confusing and messy. Maybe give a branch meaningful name (like "add feature X") and later link it with issue(s). And when you merge branch back to master, you might want to consider "rebase" strategy. 
After all - git was invented to help you understand what you did 6 months (or 6 years) ago and why.

ShpakovDmitry, 2020-02-24:
Ok, actually, i had not thought, that for non-working and unstable states it is possible to use the same feature branch. Will use this kind of flow at the moment.

P.S. Yeah, i have noticed, recently, that branch names with issue number only, are confusing and demands extra time to check what does this branch purpose really mean, which is unnecessary action. Now, i can imagine how it may be confusing if work is done on several issues at one time.
So for this purpose in last pull request a meaningful branch name was used ` coding_style_md `